SiLab-Bonn/testbeam_analysis #2: Hit masking discripancy

themperek, 2015-09-04:
This:
```python
noisy_pix_1d = (noisy_pixels[0] + 1) * (noisy_pixels[1] + 1)
pix_1d = np.empty((len(hits),), dtype=np.uint32)
np.multiply(hits['column'], hits['row'], pix_1d)
hits = hits[~np.in1d(pix_1d, noisy_pix_1d)]
```
should do same as that:

https://github.com/SiLab-Bonn/pyTestbeamAnalysis/blob/master/pyTestbeamAnalysis/analyze_test_beam.py#L73-L77

just faster but it is something else.

DavidLP, 2015-09-04:
Besides that the 1D mapping is wrong one could try that. Did you profile this to be faster? I had some bad performance expirience with np.in1d()

themperek, 2015-09-04:
Did not do profiling but much faster for me.

DavidLP, 2015-09-04:
Your mapping (col, row) --> number is not unique. Thus this it cannot work withou fixing.

themperek, 2015-09-04:
This works:
```python
noisy_pix_1d = (noisy_pixels[0] + 1) * 2**16 + (noisy_pixels[1] + 1)
pix_1d = np.empty((len(hits),), dtype=np.uint32)
np.multiply(hits['column'],  2**16, pix_1d)
pix_1d =  pix_1d + hits['row'] 
hits_1d = hits[~np.in1d(pix_1d, noisy_pix_1d)]
```

DavidLP, 2015-09-04:
I profiled it and with the example data it is about 12 % faster. With more noisy pixels it should benefit event more from calling 

hits[~noise_pixels] 
only once.

But the result is also different... This still needs to be checked where and why.

themperek, 2015-09-04:
For large data set and lot of pixels is 9134.357 ms vs 104707.988 ms
The output is for some planes equal for others not.

DavidLP, 2015-10-01:
Generic solution:

noisy_pix_1d = (noisy_pixels[0] + 1) * max_row + (noisy_pixels[1] + 1)
hits_1d = col * max_row + row
hits_1d = hits[~np.in1d(hits_1d, noisy_pix_1d)]

still need check.

DavidLP, 2015-10-27:
The solution above was correct, BUT numpy silently does the calculation 

hits_1d = col * max_row + row 

wrong if you exceed the dtype of the variables. Explicit np.astype to get correct results.

closed 2015-10-27 by DavidLP