SiLab-Bonn/testbeam_analysis #8: Fix the examples

DavidLP, 2015-11-27:


laborleben, 2016-02-21:
simulated_data_example.py gives me that error at the very last step:

2016-02-21 23:17:16,278 - root - [INFO    ] (MainThread) Calculate efficiency for DUT 0
/home/jens/miniconda/lib/python2.7/site-packages/matplotlib/tight_layout.py:222: UserWarning: tight_layout : falling back to Agg renderer
  warnings.warn("tight_layout : falling back to Agg renderer")
2016-02-21 23:17:17,855 - root - [INFO    ] (MainThread) Efficiency =  50.0000 +- 31.6228
2016-02-21 23:17:17,859 - root - [INFO    ] (MainThread) Calculate efficiency for DUT 1
2016-02-21 23:17:19,301 - root - [INFO    ] (MainThread) Efficiency =  75.0000 +- 25.0000
2016-02-21 23:17:19,306 - root - [INFO    ] (MainThread) Calculate efficiency for DUT 2
Traceback (most recent call last):
  File "/home/jens/dev/testbeam_analysis/examples/simulated_data_example.py", line 141, in <module>
    row_range=None)
  File "/home/jens/dev/testbeam_analysis/testbeam_analysis/result_analysis.py", line 330, in calculate_efficiency
    plot_utils.efficiency_plots(distance_min_array, distance_max_array, distance_mean_array, hit_hist, track_density, track_density_with_DUT_hit, efficiency, actual_dut, minimum_track_density, plot_range=dimensions, cut_distance=cut_distance, output_fig=output_fig)
  File "/home/jens/dev/testbeam_analysis/testbeam_analysis/plot_utils.py", line 809, in efficiency_plots
    fig.tight_layout()
  File "/home/jens/miniconda/lib/python2.7/site-packages/matplotlib/figure.py", line 1756, in tight_layout
    self.subplots_adjust(**kwargs)
  File "/home/jens/miniconda/lib/python2.7/site-packages/matplotlib/figure.py", line 1612, in subplots_adjust
    self.subplotpars.update(*args, **kwargs)
  File "/home/jens/miniconda/lib/python2.7/site-packages/matplotlib/figure.py", line 226, in update
    raise ValueError('left cannot be >= right')
ValueError: left cannot be >= right

themperek, 2016-02-26:
I suggest to add examples to run with tests on Travis. One could check few final parameters but even without it is useful to keep it up to date.

DavidLP, 2016-02-27:
The examples take too much time for CI. A nice way would be to reduce the example files at the beginning of a unit tests then run all the example scripts.

themperek, 2016-02-27:
This can be easily done. Just do not use all data in case TRAVIS_CI environmental variable?
One needs also to disable poping up gui.

DavidLP, 2016-02-27:
To set this unit test up takes 10 min. I would just change the example files in the unit test start up. GUI is already deactivated, but some functions need user interaction... That makes it tricky. I am planning to always provide an 'auto' mode to these functions that does not need user interaction to be able to test them.

DavidLP, 2016-04-05:
There is a unit test for the examples available and the functions do not require user interaction anymore.

DavidLP, 2016-07-15:
Unit tests are implemented but take too long and are therefore deactivated. I am not sure what to do. Data reduction does not help, since less data increases the prealignment fit time a lot.

laborleben, 2016-07-20:
I'm working right now on the EUTelescope example.
I get bad results when running the alignment. I can fix it by commenting this line:
https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/dut_alignment.py#L745
So the exiting from the loop might not give the best result.

Anyway, the best result I can achieve are these residuals:
DUT0  DUT1  DUT2  DUT3  DUT4  DUT5
  17          8         6           6         8         17      um

At least the result is symmetric. What do you expect?

DavidLP, 2016-07-22:
The residuals 17, 6, 8, 8, 6, 17 are the expected ones. 
Weird that a loop stop gives you better results since I tested that (to my mind ...). I will cross check.

laborleben, 2016-07-23:
with the latest code I can run with the original optimization loop. I improved the fitting and binning and the residuals are now the smallest I've ever seen with the EUTelescope example (improved by 0.5um, I know this is all academic :-). Also the center of the residuals are fewer away from Zero than 0.5um.

DavidLP, 2017-01-12:
Fixed.

closed 2017-01-12 by DavidLP