SiLab-Bonn/testbeam_analysis #24: Apply to events helper function

DavidLP, 2016-07-05:
Almost all multiprocessing parts apply a function to an event chunk that is aligned at the event numbers (without breaking the event number). Thus a helper function that automatically applies a function to event chunks in parallel would simplify the code and make multiprocessing easier to use.

DavidLP, 2017-01-24:
We need something like `divide, map, combine`, that does not load the arrays twice into memory. There are a lot of [multiprocessing frameworks](https://wiki.python.org/moin/ParallelProcessing) for python, but I could not find a framework easing this task. Guess we have to implement this on our own.

themperek, 2017-01-24:
http://dask.pydata.org/ ?

DavidLP, 2017-01-24:
Cannot handle pytable files.

DavidLP, 2017-09-21:
It is implemented in `map_combine` for the functions that had no multiprocessing (`hit_analysis.generate_pixel_mask`, `hit_analysis.cluster_hits`, `track_analysis.find_tracks`). The other functions will be changed when they are cleaned up after the frequent code explosions and shuffling.

DavidLP, 2017-09-21:
Implemented in `development`

closed 2017-09-21 by DavidLP