SiLab-Bonn/testbeam_analysis #30: Pixel center defined wrong

DavidLP, 2016-07-15:
Since the clusterizer was changed to give a [mean pixel index instead of a mean position](https://github.com/SiLab-Bonn/pixel_clusterizer/issues/2)  one has to add 1/2 of the x/y pixel pitch in the [cluster_hits](https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/hit_analysis.py#L108) function.

laborleben, 2016-07-15:
this is true, I have this change in the dbm branch and still needs to be merged. I will close this ASAP this is done.

DavidLP, 2016-07-15:
Perfect

laborleben, 2016-07-17:
Is it possible to use indexes still for merged cluster table and do the transition from cluster (or whatever) index to position when using apply_alignment method?

The code right now changes the pixel / cluster with the lowest index (=1, from 0.5 to 1..4999) to position 0.0 to 0.999. So the position starts at 0.0 (corner of the pixel, not center).

I'm just going to leave this here to debate.

The problem:
Hit index is an integer from 1 to max. column / max. row index. Cluster index is the ToT (or whatever) weighted index center of the clustered hits. The reason for this is to be consistent between hit index and cluster index. E.g. cluster / hit index of 1 is the center of the pixel with the lowest index. Thus the cluster / hit index ranges from 0.5 to 1.4999 for the pixel with the lowest index. At some point the transition has to be made from index to absolute position.

laborleben, 2016-07-23:
this should be it. Can you find more bugs? 👍

DavidLP, 2016-12-29:
Is this working now in development branch?

laborleben, 2016-12-29:
This should be done and all changes were committed into the development branch.

closed 2016-12-29 by laborleben