SiLab-Bonn/testbeam_analysis #32: alpha, beta correction alway positive

laborleben, 2016-07-30:
I never observed a negative number for alpha and beta correction. Thus the angle is increased every iteration and makes the residuals worse.

DavidLP, 2016-08-01:
It is mathematically only possible to determine the sign of alpha * beta. Practically maybe not even that due to finite resolution in the slope fit. Thats why I recommended to optimize the slopes to 0 with a generic minimizer.

DavidLP, 2016-08-01:
The fourth formular can be used to determine the sign, as it is also implemented [here](https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/tools/analysis_utils.py#L709).

[image]

DavidLP, 2016-08-01:
BTW: if only one angle is rotated the result is the same independent of the alpha, beta sign. It is not the goal to reconstruct correct angles, but to get angles that minimize the residuals.

laborleben, 2016-08-01:
so the minimizer should not get a alpha and beta prediction as a start value since this would bias the result. It should just guess which is the minimum residual for alpha and beta (and z) for each iteration and return the result whether it is positive or negative.

In my opinion the angles are very much under control since I fixed the binning to give good fit results. But still then, alpha and beta are always positive which gives me some worries.

DavidLP, 2016-08-01:
Sure, 0s as start values are better than wrong start values. To write this minimizer is very straight forward but I have to measure right now.