SiLab-Bonn/testbeam_analysis #35: Chi2 does not take plane rotation into account

DavidLP, 2016-08-22:
Since the planes are now rotated the chi 2 is calculated [wrong):[https://github.com/SiLab-Bonn/testbeam_analysis/blob/44eacadb6e7743b28b889258fa88d18934e9194d/testbeam_analysis/track_analysis.py#L655,]. The assumption was that the planes are not tilted which is not the case anymore. Also see #34.

This will overestimate the chi2 but should generally not give a chi 2 that is too off. Thus has to me a low priority.

laborleben, 2016-08-22:
Similar case: https://github.com/SiLab-Bonn/testbeam_analysis/blob/44eacadb6e7743b28b889258fa88d18934e9194d/testbeam_analysis/tools/plot_utils.py#L846

DavidLP, 2017-01-19:
> Similar case:

The similar case does not exist anymore. All result function properly calculate the intersections.