SiLab-Bonn/testbeam_analysis #36: Offset calculation in track_analysis.fit_tracks()

laborleben, 2016-08-22:
In track_analysis.fit_tracks(), the offset for each each track is taken from the mean of the track coordinates and is later stored in the DUT table: [here](https://github.com/SiLab-Bonn/testbeam_analysis/blob/44eacadb6e7743b28b889258fa88d18934e9194d/testbeam_analysis/track_analysis.py#L653) and [here](https://github.com/SiLab-Bonn/testbeam_analysis/blob/44eacadb6e7743b28b889258fa88d18934e9194d/testbeam_analysis/track_analysis.py#L653)

For the residual calculation in result_analysis.calculate_residuals(), the same parameters are loaded [here](https://github.com/SiLab-Bonn/testbeam_analysis/blob/44eacadb6e7743b28b889258fa88d18934e9194d/testbeam_analysis/result_analysis.py#L108) and used to calculate the residuals [here](https://github.com/SiLab-Bonn/testbeam_analysis/blob/44eacadb6e7743b28b889258fa88d18934e9194d/testbeam_analysis/result_analysis.py#L134) and [here](https://github.com/SiLab-Bonn/testbeam_analysis/blob/44eacadb6e7743b28b889258fa88d18934e9194d/testbeam_analysis/result_analysis.py#L135).

Since the offset is just the mean value of all track hits and not an intersection with the DUT plane, residuals are wrongly calculated.

DavidLP, 2016-08-23:
No, that would be totally wrong. The [fitted track is intersected with the tilted plane](https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/track_analysis.py#L278) and the correct offset is stored.

closed 2016-08-23 by DavidLP

laborleben, 2016-08-23:
Ok, I missed that.

I think it makes sense to re-calculate the intersection in result_analysis.calculate_residuals() to not depend on track_analysis.fit_tracks() results (calculate_residuals() also receives the alignment file which may change after fit tracks).

BTW what is the benefit to calculate the track-offset vector in the DUT plane and not in the (1,1,0)-plane?

DavidLP, 2016-08-23:
No I do not think so. Because every result analysis (track density, charge, residuals, etc.) would have to do the intersection calculation. That is how it was implemented in the past. It takes longer and increases the loc.

The benefit is that the Kalman filter result would look the same.