SiLab-Bonn/testbeam_analysis #37: Rotation and translation of the global coordinate system

laborleben, 2016-08-23:
I want to change the global system so that the average of all tracks goes along the z-axis/(0,0,1)-axis and the mean of the offsets through the origin (0,0,0) of the global coordinate system.

Therefore I need a rotation (and translation) of the global system. Currently, when going from local to global, we use np.dot(T, R), thus applying rotation and then translation for each DUT individually.

But this is not enough when modifying the overall position of the telescope (including all DUTs) and track.

Another set of rotation (and translation) parameters is necessary to to do this change of the global coordinate system.

Is my assumption right, that changing the global coordinate system is not possible with the current code?

DavidLP, 2016-08-23:
What you effectively want to do:

Instead of defining the _gobal coordinate system_ at the _reference DUT_, you want to define it somewhat by the _beam_.

You do not need a translation of the global system, the global system is not global if it is transformed ;-)

You can _define_ the global system different of cause (e.g. x/y = 0 at beam center). Only the local to global T and R matrices for each device will change then. Especially for DUT 0.

What is the need for this rather drastic change?

laborleben, 2016-08-23:
Yes, yes and yes :-)

This is just nicer if the system is aligned to the beam and the find_track() will work even after (fine) alignment.

But the changes are not drastic, in principle another translation and rotation around the origin which is applied after translation/ration of each device. Of course this will increase processing time but it is vectorized anyway.

We already achieve this after pre-alignment, where by definition the tracks are parallel to the z-axis. I just want to keep this geometry even after (fine) alignment.

DavidLP, 2016-08-23:
I do not get why you need another translation and rotation. Just _define_ your coordinate system do be the beam along z and change the _already existing_ matrices accordingly. 
Otherwise it would be a transformation from the _local plane coordinate system_ to the _local beam coordinate_ system with a _meaningless global coordinate_ system.

DavidLP, 2017-01-09:
@laborleben Is this clear now and I can close?

laborleben, 2017-01-09:
Some code for that is still in experimental branch. I haven't reviewed your changes, so I can't comment.

DavidLP, 2017-01-12:
> I haven't reviewed your changes yet

I have done changes?

DavidLP, 2017-01-19:
Is this still needed? The `alignment` function should align the DUTs along the beam z-axis.

laborleben, 2017-01-19:
This is still ongoing.

DavidLP, 2017-01-23:
What is ongoing? This feature is theoretically there.

DavidLP, 2017-04-04:
Closed due to inactivity.

closed 2017-04-04 by DavidLP