SiLab-Bonn/testbeam_analysis #39: Regressions in development branch

DavidLP, 2016-12-29:
The development branch suffered a lot of regressions and cannot be used productively right now. There is an experimental branch, but the functions there are work in progress, have a lot of debug prints, and are not tested.

The following points have to be fixed:

- [ ] Examples have to work again
- [ ] Results have to be cross checked with later analysis
- [ ] Merging of cleaned up experimental into development has to be done. Right now development is a messy, not working mixture of both (e.g. efficiency calculation does not work)

DavidLP, 2017-01-09:
Cross check with old data (FE-I4 telescope + LFoundry sensor) and simulation show the same efficiency and similar # tracks etc. Thus results can be trusted again.

DavidLP, 2017-01-09:
Simulation example and FE-I4 telescope example do work again with good results.

DavidLP, 2017-01-12:
EUtelelecope data example works with good results. For reference:
- Residuals after prealignment:
``` 
column = [21.5 ; 7.9; 17.2; 17.8; 17.7; 36.0] um
row = [19.8; 6.9; 21.9; 21.7; 22.9; 50.9] um
```
- Residuals after alignment with all DUTs in fit and Chi2 track cut: 
``` 
column = [17.1; 6.3; 8.2; 8.0; 6.4; 17.3] um
row = [17.1;  6.3; 8.2; 8.1; 6.3; 17.2] um
``` 
- Residuals after alignment with neighbouring DUTs in fit: 
``` 
column = [12.9; 6.5; 6.5; 6.4; 6.6; 13.1] um
row = [12.8; 6.5; 6.6; 6.4; 6.5; 13.0] um
```

DavidLP, 2017-01-12:
All Examples with shipped data have unit tests now.

DavidLP, 2017-01-12:
@laborleben Code is in good health now and was released. Feel free to merge your experimental branch now.

DavidLP, 2017-01-19:
@laborleben : I compared `development` to `experimental`. I do not see that much of a difference in features:

1. Some fitting limits in `prealignment` that are used in `calculate_residuals`
2. Unifinished refactoring of `calculate_efficiency`. What does it do?

Mayor differences:
- `track_quality` definition is different
- `dut_alignment.alignment` refactoring or are also features added?

I would just cherry pick some lines of code of `experimental` for `development` and then abandon `experimental`. Ok?

laborleben, 2017-01-19:
The branch needs a rebase, because you did some changes that produces conflicts. I will clean up the experimental branch and check for errors again.

laborleben, 2017-01-24:
Most of the smaller things were merge into the development branch. I will create another feature branch with the important changes from the experimental branch.

DavidLP, 2017-01-24:
I can plot charge maps now?

DavidLP, 2017-03-16:
The status of the development branch is OK right now. Thus I close.

closed 2017-03-16 by DavidLP