SiLab-Bonn/testbeam_analysis #46: Result analysis: Track angle

DavidLP, 2017-01-23:
To check the overall alignment a beam angle theta histogram would be useful.

YannickDieter, 2017-01-24:
Okay. To add this function I will create a new branch 'new_features' okay? Should I create one histogram for all DUTs or better separated histograms for the different DUTs? Also a RMS and MEAN calculation would be useful?

DavidLP, 2017-01-24:
Tracks are build for a certain device (with different cuts) from the tracks tables. So it makes sense to calculate the track angles for each device, similar how it is done right now for the other [result analysis](https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/result_analysis.py#L20) (parameter `use_duts`). I would call the new branch `track_angle`, or are you planning to add other features?

YannickDieter, 2017-01-24:
Okay this makes sense. No, but maybe we add more features later.

DavidLP, 2017-01-24:
> No, but maybe we add more features later.

Then we create a new branch ;-)

YannickDieter, 2017-01-25:
I pushed the track angle histogram. Is it better to separate calculation and plotting even for this 'small' function? I also created a .h5 file with the fitted means and sigmas from the track angle histogram.

DavidLP, 2017-01-25:
> I also created a .h5 file with the fitted means and sigmas from the track angle histogram.

Good.

We should always seperate calculations from plotting. This makes a seperation between plotting library / backend etc. from calculations possible.

laborleben, 2017-01-25:
Separating plotting from the other is probably better. Keep in mind that you need to do all the additional imports otherwise.

DavidLP, 2017-01-25:
Created a [PR](https://github.com/SiLab-Bonn/testbeam_analysis/pull/47)

YannickDieter, 2017-01-25:
> We should always seperate calculations from plotting. This makes a seperation between plotting library / backend etc. from calculations possible.

Okay then I will split plotting and calculation.

DavidLP, 2017-01-25:
Also you could merge the changes from development frequently, since Jens is changing it right now.

DavidLP, 2017-02-07:
This feature is in development now.

closed 2017-02-07 by DavidLP