SiLab-Bonn/testbeam_analysis #51: Regression in prealignment plotting

DavidLP, 2017-02-07:
Offset and error cuts are not plotted anymore:

[image]

laborleben, 2017-02-07:
The errorbars were removed because they filled up the complete screen several times. This made the plot completely useless.
Anyway, what kind of information do you expect to get in addition from the errorbars? They were also misleading in double Gaussian fit several times ( bad fit). So the offset was enough to select a good range. The precision is anyway good enough for the following alignment.

laborleben, 2017-02-07:
They don't need to be plotted in this case because the fit range selects the data between the two red bars and the plot zooms in as much as possible with the maximum error/offset at the top of the plot.

DavidLP, 2017-02-07:
> fit range selects the data between the two red bars

Which red bars? The blue red bars? ;-)

I actually had the impression that no matter what value I put for the error/offset cut, they never show up. I cross check.

closed 2017-02-09 by laborleben