SiLab-Bonn/testbeam_analysis #52: [tools.plot_utils] Regression in figure saving

DavidLP, 2017-02-07:
There was a clean up of the plotting tools in commit b78b4d95eaf3dc028f16fb1d2abc3f054115e976 where the fig argument was dropped in `output_pdf.savefig(fig)` for figures not managed by matplotlib. This leads to a crash under windows over remote desktop. But the unittests do not fail.

``` bash
  File "C:\Users\DavidLP\Miniconda2\lib\site-packages\matplotlib\backends\backend_pdf.py", line 2469, in savefig
    raise ValueError("No such figure: " + repr(figure))
ValueError: No such figure: None
```

DavidLP, 2017-02-07:
Shows how fishy the figure handling is. I guess best practice is to always give the figure explicitly and do not trust matplotlib to provide the correct figure handle.

laborleben, 2017-02-07:
usually no argument needed. This is stated several times in the matplotlib 2.0 documentation:http://matplotlib.org/api/backend_pdf_api.html#matplotlib.backends.backend_pdf.PdfPages.savefig

Certainly not ideal, and as I stated before, plotting crashes in screen. I'd be happy to fix this once and  for all.

DavidLP, 2017-02-07:
It says that the active figure is plotted, but I think there is no active figure?

DavidLP, 2017-02-07:
For the interface we have to change the plotting library nevertheless quite a lot

YannickDieter, 2017-02-08:
Have the same error as David reported in calculating the efficiency.
``  
File "C:\Users\DavidLP\Miniconda2\lib\site-packages\matplotlib\backends\backend_pdf.py", line 2469, in savefig
    raise ValueError("No such figure: " + repr(figure))
ValueError: No such figure: None 
``

laborleben, 2017-02-08:
Most of the plotting code relies on pyplot, which can make trouble when using threaded code and when no GUI is available (screen, CI, ...).

We should change to the following pattern:
```
from matplotlib.backends.backend_pdf import PdfPages
from matplotlib.backends.backend_agg import FigureCanvas

```
from matplotlib.figure import Figure

output_pdf = PdfPages("test1.pdf")

fig = Figure()
canvas = FigureCanvas(fig)
ax = fig.add_subplot(111)
ax.plot([1,2,3], [2,3,4])
ax.set_title("Title")
output_pdf.savefig(fig)
canvas.print_figure('test2.pdf')
fig.savefig("test3.pdf")
```

DavidLP, 2017-02-08:
Yes. And for the ability to still plot something to screen the following works:

``` python
from matplotlib.backends.backend_pdf import PdfPages
from matplotlib.backends.backend_agg import FigureCanvas

from matplotlib.figure import Figure

def create_plot(fig=None):
  if not figure:
    fig = Figure()
  canvas = FigureCanvas(fig)
  ax = fig.add_subplot(111)
  ax.plot([1,2,3], [2,3,4])
  ax.set_title("Title")
  output_pdf.savefig(fig)
  canvas.print_figure('test2.pdf')

def show_plot():
    ''' Create figure managed by matplotlib to allow plot to screen '''
    fig = plt.figure()
    create_plot(fig)
    plt.show()
```

laborleben, 2017-02-09:
This should be fixed. See #54.

commit 460bb37f (2017-02-08): ENH: clean up plotting, partially fixes #52
commit 6087a70b (2017-02-08): ENH: fix plotting, partially fixes #52
commit f83d6917 (2017-02-09): ENH: fix plotting, partially fixes #52
closed 2017-02-09 by laborleben