SiLab-Bonn/testbeam_analysis #55: Provide data files externally to python package

DavidLP, 2017-02-09:
To run unit tests and provide example data one needs to have rather large data files. These files should and cannot (max. quota ~ 250 MB) be stored in the python package hosted on pypi. I did not find a good solution to do this. The best would be a python package providing autodowload of files on demand for several supported file hosters. Otherwise github could be misused to a certain extend, by adding a data branch and downloading from there...

DavidLP, 2017-02-17:
We have 150 Mb per branch (because of the data files) and 6 branches = 6 * 150 = 900 Mb. Github allows for 1 Gb before you get a "polite" E-Mail. Thus one more branch is not supported anymore and we have to find a solution asap. I will create a data branch within testbeam analysis and add a downloader tool for the exampels.

DavidLP, 2017-05-22:
Did not come up with a good solution yet. Hosting at University is not a trivial task and you do not want to start username/psw access restrictions.

DavidLP, 2017-07-18:
https://quiltdata.com/ look promising

DavidLP, 2017-09-19:
Implemented in development

DavidLP, 2017-12-04:
The Scibo campuscloud does not seem to be planned as a high availability cloud, frequent downtime of up to days (when it is weekend) are common. Additionally the downtime (an important classifier for an online hoster) seems to be disclosed. This does not meet our requirementa, since the downtime stalls our tests and builds. We should search for a better service. I reopen this issue.

themperek, 2017-12-04:
https://help.github.com/articles/distributing-large-binaries/

DavidLP, 2017-12-05:
This seems to be exactly what we want, even in same repo: https://git-lfs.github.com/

laborleben, 2017-12-05:
This is maybe the only option.