SiLab-Bonn/testbeam_analysis #57: [dut_alignment] Crash in alignment for example

DavidLP, 2017-02-09:
The `telescope_time_reference` crashes now:

``` bash
2017-02-09 12:46:07,190 - root - [INFO    ] (MainThread) === Calculating residuals ===
2017-02-09 12:46:07,200 - root - [INFO    ] (MainThread) Use alignment data
C:\Users\DavidLP\Miniconda2\lib\site-packages\scipy\interpolate\fitpack.py:731: RuntimeWarning: The number of zeros exceeds mest
  warnings.warn(RuntimeWarning("The number of zeros exceeds mest"))
Traceback (most recent call last):
  File "C:\Users\DavidLP\git\testbeam_analysis\testbeam_analysis\examples\telescope_time_reference.py", line 247, in <module>
    run_analysis()
  File "C:\Users\DavidLP\git\testbeam_analysis\testbeam_analysis\examples\telescope_time_reference.py", line 169, in run_analysis
    pixel_size=pixel_size)
  File "C:\Users\DavidLP\git\testbeam_analysis\testbeam_analysis\dut_alignment.py", line 936, in alignment
    chunk_size=chunk_size)
  File "C:\Users\DavidLP\git\testbeam_analysis\testbeam_analysis\dut_alignment.py", line 981, in _duts_alignment
    chunk_size=chunk_size)
  File "C:\Users\DavidLP\git\testbeam_analysis\testbeam_analysis\dut_alignment.py", line 1059, in _calculate_translation_alignment
    chunk_size=chunk_size)
  File "C:\Users\DavidLP\git\testbeam_analysis\testbeam_analysis\result_analysis.py", line 438, in calculate_residuals
    output_pdf=output_pdf
  File "C:\Users\DavidLP\git\testbeam_analysis\testbeam_analysis\tools\analysis_utils.py", line 857, in fit_residuals_vs_position
    fit, cov = curve_fit(linear, xcenter[x_sel], y_mean[x_sel], sigma=y_rel_err[x_sel], absolute_sigma=False)
  File "C:\Users\DavidLP\Miniconda2\lib\site-packages\scipy\optimize\minpack.py", line 676, in curve_fit
    res = leastsq(func, p0, Dfun=jac, full_output=1, **kwargs)
  File "C:\Users\DavidLP\Miniconda2\lib\site-packages\scipy\optimize\minpack.py", line 380, in leastsq
    raise TypeError('Improper input: N=%s must not exceed M=%s' % (n, m))
TypeError: Improper input: N=2 must not exceed M=1
```

laborleben, 2017-02-09:
I have an idea, but the crash log is not helpful. I don't have your data to repeat the analysis. Can you provide the data?

DavidLP, 2017-02-09:
Sure. It is on `sirrush/silab/david/jens/example_data`. I already put noise free cluster files since you do not want to repeat these steps (takes over 30 min.).

laborleben, 2017-02-13:
You are requiring hits in DUT3 for DUT4, but DUT3 is rather small and is positioned outside the fit limit range.

2 solutions:
- do not use fit limits while aligning the DUTs (`use_fit_limits=False`)
- do not require a hit in DUT3 for DUT4 (and set `use_fit_limits=True`)

In both cases, alignment should be possible and the output is OK.

DavidLP, 2017-02-13:
> do not require a hit in DUT3 for DUT4 (and set use_fit_limits=True)

This is already the case. No hit in DUT3 is required when fitting for DUT4. Check [here](https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/examples/telescope_time_reference.py#L139).

laborleben, 2017-02-13:
Your parameters are effectively removing all tracks.

See here: https://github.com/SiLab-Bonn/testbeam_analysis/blob/39ee16b38256bacaa163329854a79fbda41d0298/testbeam_analysis/dut_alignment.py#L975-L975

Can you try:
```
align_DUTS=[[3]]
selection_fit_duts=[0,1,2,5,6,7]
selection_hit_duts=[0,1,2,5,6,7]
selection_track_quality=[1,1,1,1,1,1]
```

DavidLP, 2017-02-13:
Why should my parameters remove all tracks? It did not in the past and should not do this now. The selections in the examples are unchanged and still correct. You want to have:

```python
align_DUTS=[[3]]
selection_fit_duts=[0,1,2,5,6,7]
selection_hit_duts=[0,1,2,3,5,6,7]
selection_track_quality=[1,1,1,1,1,1]
```
Otherwise you do have too less tracks that also have a DUT3 hit for alignment.

laborleben, 2017-02-13:
You are right, the behavior has changed because there were severe bugs that affected the multi-DUT runs: 9feaa87c0f051b3dadb4df3896aeaee7f0822f13 and 75caa5133f41f38bcdb23b2d2cf619a4dd971bf5.
The initial track selection quality which you provide for DUT3 (`selection_track_quality=[1,1,1,1,1,1]`) will not be met by any track because there is no initial alignment (transformation matrix is unity matrix).
For a fix use the parameters which I provided or use the [improvements](SiLab-Bonn/testbeam_analysis/tree/improvements) branch where initial parameters are taken from the pre-alignment.

laborleben, 2017-02-13:
Also set
```Python
dut_alignment.alignment(use_fit_limits=False)
```

laborleben, 2017-02-13:
Even if you use initial alignment parameter, remember that we changed the coordinate system. I don't know if the initial parameters are from before or after the change.

DavidLP, 2017-02-13:
Reopen because that is a not fixed regression since the example and thus the development branch still do not work in such setups. It is fixed if the example works again.

> The initial track selection quality which you provide for DUT3 (selection_track_quality=[1,1,1,1,1,1]) will not be met by any track because there is no initial alignment (transformation matrix is unity matrix).

I cannot follow. The track quality is set in track finding and has nothing to do with a transformation matrix. This selection is independent of any alignment and should still work. Also there is an initial alignment, it is given as a parameter?!

laborleben, 2017-02-13:
Yep, track quality is set up before. This wasn't the issue.
Anyway, the example works, with the parameters given in the example file and with the latest development branch.

DavidLP, 2017-02-14:
I check

DavidLP, 2017-02-14:
No crash anymore in alignment function.

closed 2017-02-14 by DavidLP