SiLab-Bonn/testbeam_analysis #58: [dut_alignment] Cannot change error / offset cut in interactive pre-alignment

DavidLP, 2017-02-14:
When I click on the sliders nothing happens.

laborleben, 2017-02-14:
This is working on my machine. Can you give a more detailed bug report?
The click may have a short delay because it triggers a refit. Probably you are running heavy jobs in the background and the delay becomes noticeable.

laborleben, 2017-02-14:
I notice a delay when I run heavy job, but this is in the order of 1s. The usual delay is much shorter.

DavidLP, 2017-02-14:
> This is working on my machine. Can you give a more detailed bug report?

Take the Mimosa telescope example, change to interactive prealignment, click more than once on error/offset limit slider:

[image]

It stops responding,

> I notice a delay when I run heavy job, but this is in the order of 1s. The usual delay is much shorter.

Nothing was running besides this. First change was done after ~3s. Then no changes anymore. CPU is idle. So load is not a reason.

commit e3c1c998 (2017-02-14): BUG: fix broken slider, check for at least 2 valid data points in selected range, fixes #58
closed 2017-02-14 by laborleben