SiLab-Bonn/testbeam_analysis #59: [dut_alignment] Bad results for example: Mimosa26 + time reference

DavidLP, 2017-02-14:
The alignment of the planes fail in the example. The residuals are much too large.

Comparison to before:

| **DUT** | **Before**     | **After**        |
|-----|------------|--------------|
|  0  |   4.5/4.4  |   4.5/26.9   |
|  1  |   4.6/4.6  |   4.2/12.6   |
|  2  |   4.5/4.4  |   4.2/10.0   |
|  3  |  90.5/17.1 |  94.9/375.5  |
|  4  | 100.5/15.8 | 175.9/1519.8 |
|  5  |   4.4/4.3  |   4.1/24.8   |
|  6  |   4.5/4.4  |   4.2/19.7   |
|  7  |   5.2/4.8  |   4.5/45.7   |

Col/Row residuals in um.

DavidLP, 2017-02-14:
Also efficiency calculation crashed, but this might be related to bad alignment.

laborleben, 2017-02-14:
I guess you are referring to the development branch?!

The alignment algorithm wasn't changed. So I don't have an explanation yet.

laborleben, 2017-02-15:
What is the "before"? Which commit hash?

DavidLP, 2017-02-15:
Master branch

DavidLP, 2017-02-15:
Ok I check with master again

DavidLP, 2017-02-15:
Master is already failing ...

DavidLP, 2017-02-15:
@laborleben A candidate why this is worse now is your commit **https://github.com/SiLab-Bonn/testbeam_analysis/commit/2574efd749b275760db961916eeb7ca4d0fa6cea#diff-68995e26f93a03bd15d48b3135026068**, where you revert the `def fit_residuals()` function to an older solution. I was using `scipy.stats.binned_statistic_2d`. Interesstingly github cannot compare these changes, was a `git rebase + git push force` involved? Why do you revert to an older solution, did you do tests? I doubt that fitting these 2.5 bins now makes sense:
[image]

DavidLP, 2017-02-15:
Also you commit message does not mention anything related to this rather drastic change.

laborleben, 2017-02-15:
1. binned_statistics was using only 1 data chunk while lopping over data table. This was fixed by binning the data while looping over it.
2. The fits were more robust when using a coarse binning for the first alignment step. In the current master branch (index) the first 3 iterations are using a coarse binning which was reverted in development to use no coarse binning at all (7a5b25a15069c627228f151f6144e75f1a5de7b8).

DavidLP, 2017-02-16:
After Fix:

| **DUT** |  **After fix**        |
|-----|------------|
|  0  |   4.5/4.4  |  
|  1  |   4.2/4.3  | 
|  2  |   4.2/4.1  |
|  3  |  90.8/17.5 |
|  4  | 100.6/15.6 |
|  5  |   4.1/4.1  |
|  6  |   4.2/4.2  |
|  7  |   4.5/4.5  |

The performance for the Mimosa planes is actually now up to 12% better! For the small device slightly worse.

closed 2017-02-16 by DavidLP

laborleben, 2017-02-16:
well spotted!

DavidLP, 2017-02-16:
THX. This bug was a `Scydosella musawasensis`. Almost too tiny to find.

commit af3bb1bb (2017-02-16): REG: fix #59