SiLab-Bonn/testbeam_analysis #60: [dut_alignment] Binning in residuals vs. position hists

DavidLP, 2017-02-16:
The binning shows very strong ´Moiré effects and should be optimized:
[image]

laborleben, 2017-02-16:
There is the possibility to adjust the binning: https://github.com/SiLab-Bonn/testbeam_analysis/blob/af3bb1bb7b70e5bd8b097baf8d341c1d3a843030/testbeam_analysis/dut_alignment.py#L1080.
The coarse binning helps to improve the fit. The effect the plot is showing is by far not the worst.

The effect is probably both, a cobination of a binning effect (plotting) and a pixelized telescope and DUT.

DavidLP, 2017-02-17:
If the fit result depends on the binnig then this is not correct and missuses the bin width as a frequency filter. Also I do not want to optimize bin width by hand for each device. Choosing the "best" bin size is a science itself ([paper](https://arxiv.org/pdf/physics/0605197.pdf)) and I was hoping `scipy.binned_stat` does that, but apparently only `np.histogram` has fancy algos for autobinning,

laborleben, 2017-02-17:
We use "auto" binning if no binning value is given. The "auto" binning is part of numpy 1.11.

DavidLP, 2017-02-17:
And `auto` uses `Freedman Diaconis Estimator` and `sturges`  auto bin method , where I guess somebody thought about that.

DavidLP, 2017-02-17:
Sorry Firefox missclicked