SiLab-Bonn/testbeam_analysis #61: Mean efficiency calculation

DavidLP, 2017-04-04:
The mean efficiency calculation should weight the mean result by the statistics per bin. Otherwise low statistic bins can change the mean efficiency a lot (e.g. imagine only one track in a bin creates either 100% or 0% efficiency). Right now this can be mitigated by requiering a minimum amount of tracks per bin and this number should be set always.

laborleben, 2017-04-06:
https://de.wikipedia.org/wiki/Will-Rogers-Phänomen 👍

DavidLP, 2017-04-06:
If you do not give a number on the error of the mean this can happen.

owtscharenko, 2017-04-06:
would this be solved by weighting single bins with the sum of total tracks and use of a weighted average?

laborleben, 2017-04-06:
When you mean wighted by the total tracks of each individual bin, then yes, but it is not yet done.

DavidLP, 2017-04-06:
It is less lines of code to implement this than we wrote here :-D. So far we calculate the mean effiencies of the pixels not of the sensor (yes that is something different). If we want to have the overall efficiency, one can just do:

```python
total_track_density_with_DUT_hit.sum() / total_track_density.sum()
```

DavidLP, 2017-06-19:
@owtscharenko Did you add any calculation in your code that is worth uploading?

DavidLP, 2018-04-24:
So found a proper solution that does not require ROOT.

DavidLP, 2018-05-25:
Implemented

closed 2018-05-25 by DavidLP