SiLab-Bonn/testbeam_analysis #63: [GUI] Enhancements

DavidLP, 2017-05-10:
Enhancements for the different tabs (no bugs, for this other issues!)
- [ ] Files: Drag and drop area for data files
- [ ] Setup: Show set angles x 10 in plot and correct relative distances
- [ ] All tabs: Fixed parameter lists should be shown with a new line for each parameter for readability
- [ ] Alignment: Should be optional but not mandatory
- [ ] Track fitting: Ability to use pre-aligned tracks, since Alignment step should not be mandatory
- [ ] Result: Should be split into optional result analysis steps (residuals, efficiency, ...)
- [ ] All tabs: Ability to open created h5 files(s) with ViTables

leloup314, 2017-05-13:
`Traceback (most recent call last):
  File "/home/leloup/miniconda2/bin/vitables", line 37, in <module>
    from PyQt4 import QtGui
ImportError: No module named PyQt4
`
ViTables does not support PyQt5 but only PyQt4; cannot use ViTables with PyQt5

DavidLP, 2017-05-13:
True. That's why I recommended to just issue the command `vitables file.h4` in a console process when clicking the button:
```python
from subprocess import call
call("vitables file.h4")
```
 This allows to start system installations of *vitables* and *vitables* installations in other environments.

leloup314, 2017-05-13:
[image]
Like this? This does not work for me even though the file exists in the cwd

DavidLP, 2017-05-13:
Do `sudo apt-get install vitables` first. Should work

leloup314, 2017-05-17:
I think showing correct relative distances is not useful, since it can hardly be seen. I would suggest to stick to equidistant separation.

leloup314, 2017-05-18:
Is the material budget a DUT property? What's the unit and how is it different from the thickness?

DavidLP, 2017-05-18:
> Is the material budget a DUT property?

Yes

> What's the unit

@YannickDieter You did the calculation recently. Can you answer? I guess it is g / cm³

> how is it different from the thickness

You can have different materials and the material budget also accounts for passive material (PCB, readout) etc.

YannickDieter, 2017-05-18:
Yes but for the Kalman Filter we need matrial budget (radiation length) in um. So you need to divide it by the density of the material.

YannickDieter, 2017-05-18:
At the moment, the kalman filter needs for every DUT the whole thickness (sensor + other materials) and the corresponding radiation lengths in um for the material (or compound).

YannickDieter, 2017-05-18:
>  @YannickDieter You did the calculation recently. Can you answer? I guess it is g / cm³

sorry, it is g / cm². This gives cm if you divide it by the density.

leloup314, 2017-05-18:
So you would need not only a material budget input but also a one for the density and radiation length of the material?
Or should the correctly calculated quantity be put in directly?

YannickDieter, 2017-05-18:
I dont know how we should do this in the future. But at the moment I only need radiation length (which is the same as material budget for me) in um + the whole thickness. So the user has to calculate the radiation length. Maybe it is better to this automatically, but then we need to pass the matrial as well.

leloup314, 2017-05-18:
As far as I understood, we also want to consider two different materials e.g. sensor and PCB or housing. Does the Kalman filter take these also into account e.g. we would need: thickness_sensor + X_0_sensor + thickness_PCB + X_0_PCB?

YannickDieter, 2017-05-18:
No not really. But e.g. for M26 (where I consider different materials: 50 um Si + 50 um Kapton), the user should take 100 um as total thickness and a radiation length of 125390 um. This follows from this formula: https://cds.cern.ch/record/1279627/files/PH-EP-Tech-Note-2010-013.pdf (radiation length of compounds)

leloup314, 2017-05-18:
Okay, so we need one field for total thickness and one for the overall radiation length?

YannickDieter, 2017-05-18:
Yes, but for each DUT ;)

leloup314, 2017-05-18:
[image]
This occurred while testing. Does anyone know why?

YannickDieter, 2017-05-19:
I also had this problem when calculating the residuals of the eutelescope example for Kalman tracks. The residuals for the first plane were all very close to zero (I think I know the reason now) and then the histogram function causes the error since if no bins are specified `nbins` is set to `'auto'`. Specifying `npixels_per_bin, nbins_per_pixel` in the residual calculation should fix it.

DavidLP, 2017-05-19:
Added a new issue. Related observations please add there. Also please always report issues, spotting and describing bugs is halve the work!

leloup314, 2017-05-20:
[image]
[image]
These two errors have been occurring quite some times during testing. Especially when having mixed DUT types e.g. one FE-I4 and rest M26

DavidLP, 2017-05-22:
This is another bug. Please open a bug report with info how to reproduce it.

DavidLP, 2017-05-29:
@YannickDieter Can you upload your scattering plane track fitting, that we can see the API?

YannickDieter, 2017-05-29:
Do you mean the option to add a additional scattering plane? Is it possible to push again into developement?

leloup314, 2017-06-16:
@YannickDieter Do we really need an `index_scatter` for the scatter planes since we have the z positions which determine the index already, I guess. See [here](https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/track_analysis.py#L196-L203)
Is it possible to process more than one scatter plane? Right now one can add multiple scatter planes in the setup.

leloup314, 2017-06-16:
@DavidLP @YannickDieter  Should we add an input parameter [`beam energy`](https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/track_analysis.py#L191-L192) to the setup tab?

DavidLP, 2017-06-17:
I think `index_scatter` should be removed.

leloup314, 2017-06-17:
Is it implemented in `cluster_hits` function to have an `input_noisy_pixel_mask_file` which does not have a mask for all DUTs e.g. 4 DUTs and only 1 noisy pixel mask for one of the DUTs in input_noisy_pixel_mask_file?

YannickDieter, 2017-06-19:
For support of multiple scattering planes and changes related to this option, see #78.

leloup314, 2017-08-16:
@DavidLP should I keep track of every enhancement here or just close?

DavidLP, 2017-08-17:
Looks good. Is the last point implemented?

leloup314, 2017-08-17:
Well, it's not yet implemented but I'm asking in general. Should I keep this list updated with every enhancement I implement or should I at some point close this 'issue' and resume to announce the enhancements just in the commit messages?

DavidLP, 2017-08-18:
I would say we summarize all needed features for the first release here. For other not mandatory features we can create a new issue then.

DavidLP, 2017-08-21:
I think drag and drop does not work for the file selection?

leloup314, 2017-08-21:
Drag and drop works since May (678fae6d1d40d77db9b93258632f3a091f1feb59). I'm also very sure I showed you this feature then.

DavidLP, 2017-08-21:
True, thought it was not working on Friday, but when I checked on Windows now it was working.

DavidLP, 2017-08-30:
This should be just a tick box for every DUT, no?
[image]

leloup314, 2017-08-30:
Yes, this should be just tick boxes, but I haven't changed this yet, since this is handled in [here(https://github.com/SiLab-Bonn/testbeam_analysis/blob/gui/testbeam_analysis/gui/option_widget.py#L187) and I only fixed the option widget for multiple boxes. This should work for now

leloup314, 2017-08-30:
Set as fixed parameter in 6c251b9b1253d137bb261050ea8d6df7a304c5a8

leloup314, 2017-09-08:
Fixed in 35b57e082ca69befdb474ee12c0d9b8797c7e57e

leloup314, 2017-09-08:
@DavidLP Plotting is now done for all tab (except alignment, since I dont know what to plot there). Please have a look at 67e34132d6f67f6b4e577c38605f6326fd2aa2da. I changed `plot_utils` so it doesn't cointain code dublicates anymore, overall I had to change several files slightly to enable plotting of `prealignment`, `efficiency` and `residuals` since there are no standalone plotting functions for them and to implement them doesn't seem trivial at all. Does this look okay to you?