SiLab-Bonn/testbeam_analysis #66: [development] test_track_finding.py fails when when not jitted

laborleben, 2017-05-17:
Removing the `@njit` decorator from `_find_tracks_loop` (https://github.com/SiLab-Bonn/testbeam_analysis/blob/79a4857728a0259369d0ae8c4c6bdfe757f05523/testbeam_analysis/track_analysis.py#L590) will lead to a failing unit test:
```
Traceback (most recent call last):
  File "/home/jens/dev/testbeam_analysis/testbeam_analysis/testing/test_track_analysis.py", line 56, in test_track_finding
    chunk_size=293)
  File "/home/jens/dev/testbeam_analysis/testbeam_analysis/track_analysis.py", line 113, in find_tracks
    _find_tracks_loop(tracklets_data_chunk, tr_x, tr_y, tr_z, tr_xerr, tr_yerr, tr_zerr, tr_charge, column_sigma, row_sigma, min_cluster_distance)
  File "/home/jens/dev/testbeam_analysis/testbeam_analysis/track_analysis.py", line 662, in _find_tracks_loop
    _set_dut_track_quality(tr_x, tr_y, track_index, dut_index, actual_track, actual_track_column, actual_track_row, actual_column_sigma, actual_row_sigma)
UnboundLocalError: local variable 'actual_track_column' referenced before assignment
```

DavidLP, 2017-05-18:
Weird.

DavidLP, 2017-05-18:
Will look into it when I find time

DavidLP, 2017-05-23:
Fixed.

closed 2017-05-23 by DavidLP