SiLab-Bonn/testbeam_analysis #69: Alignment crashes due to auto scaling

leloup314, 2017-05-23:
[image]
When using the 4 FE-I4 hit files from the example data, an consecutive analysis like in the `eutelescope` example (with default values) can be done until the alignment crashes with the above ValueError.

DavidLP, 2017-05-24:
The real issue in your case is that there are no tracks at all to plot. But nevertheless the plotting function should not crash for no data. Fixed.

leloup314, 2017-06-18:
This still occurs for alignment

DavidLP, 2017-06-19:
Ok. We should check all plotting functions that the empty data case when using a log axis is handled.

commit ddf37ecb (2017-05-24): BUG: fixes #69