SiLab-Bonn/testbeam_analysis #70: Pre-alignment crashes for different DUT types when fitting

leloup314, 2017-05-23:
[image]
The pre-alignment crashes whenever there are different DUTs, e.g. 1 FE-I4 and the rest M26 etc.
It does work for FE-I4 only and respectively M26 only. All analysis function before (like in `eutelescope` example) run.

DavidLP, 2017-05-24:
Do you have data with FE-I4 and Eutel? This is not shipped with TBA.

DavidLP, 2017-05-24:
This is not a bug, since your try correlating uncorrelated data. But I think that the error message should be more clear. Something like: 'WARNING: No correlation found, cannot fit' and then continue with the other devices.

leloup314, 2017-09-22:
@laborleben Has this been adressed in the `improvements` branch?

DavidLP, 2018-05-25:
Done

closed 2018-05-25 by DavidLP