SiLab-Bonn/testbeam_analysis #72: [GUI] Bugs

DavidLP, 2017-05-25:
This issues summarizes all parts that do not work as intended in the GUI. Checked bugs are fixed

- [ ] Crash of the `python.exe` under Windows when doing Noisy Pixel analysis with FE-I4 data
- [ ] Unit tests fail

DavidLP, 2017-05-25:
Since GUI a new feature and not the usual way to use TBA I label this an enhancement.

leloup314, 2017-06-17:
All tracebacks and error outputs should be fixed with ec450aa441ba7f9780f00ec27425d64f2977b7e6

leloup314, 2017-06-18:
The plot for tracks is made from the `TrackCandidates_prealigned.h5` after first doing `find_tracks`. I just put this there temporarily, I also do not understand waht the parameter `dut` of type `uint` in function [`plot_events`](https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/tools/plot_utils.py#L619) is supposed to accomplish, since I thought that we want to plot tracks in at least several DUTs but not only one. Maybe one can not use this immediately after `find_tracks`

DavidLP, 2017-06-19:
We are fitting tracks for a certain DUT, with independent cuts. This parameter should just select the tracks we fitted for a certein DUT.

leloup314, 2017-06-28:
Crash of python.exe is most likely due to non-threadsafe behavior of the logging handler + streamer object. @DavidLP Please check if GUI still crashes if you comment out [this line](https://github.com/SiLab-Bonn/testbeam_analysis/blob/gui/testbeam_analysis/gui/main.py#L84) in `main.py`.

DavidLP, 2017-06-29:
Unfortunately is does not help and still crashes.

leloup314, 2017-06-29:
I have to check, I guess the logging from other than main thread has to be dealt with, even though it's not redirected to a QWidget, it fails because it logs from other than main thread to console of Eclipse etc. I will reimplement analysis without multi-threading to check that this is the problem.

leloup314, 2017-07-04:
@DavidLP can you check whether `python.exe` still crashes under Windows for `THREADING_MODE == 'Pool'`? Is already default in latest commit. Thank you!

DavidLP, 2017-07-04:
It does not crashing anymore, but the GUI is not responding during analysis.

leloup314, 2017-07-04:
It was not supposed to be responding during analysis since this is not a multi-threading approach but only a reimplementation (or regression) to check whether the crashing of python has something to do with multi-threading under windows. I will try to implement something thread-safe for windows regarding logging from other than main thread

leloup314, 2017-07-24:
Crash of python.exe under Windows due to ParallelAnalysisWidget and the way threads are accessed in it.

DavidLP, 2017-10-03:
Since this is now in `development` bug are considered bug.

leloup314, 2017-10-05:
Windows and Linux handle QThreads differently. Each Qt application (GUI) has a global QThreadPool which handles all QThreads ([see here](http://pyqt.sourceforge.net/Docs/PyQt4/qthreadpool.html)) Creating instances of QThread in a loop under Windows causes the GUI to crash due to different settings of the global QThreadPool (e.g. maxThreadCount), I guess. Better approach would be to use `QThreadPool` and `QRunnable` for GUI-multithreading, which enables an automatic handling of threads independent from the OS.

DavidLP, 2017-10-06:
Go for it ;-)

leloup314, 2017-10-06:
What are the circumstances of:
`Traceback (most recent call last):
  File "main.py", line 649, in run_consecutive_analysis
    if self.tw[self.current_analysis_tab()].analysis_thread.isRunning():
RuntimeError: wrapped C/C++ object of type QThread has been deleted
Aborted (core dumped)`
`CTRL-R` works for me when I just press it.

DavidLP, 2017-10-06:
After everything ran I pressed it.

leloup314, 2017-10-07:
Okay, fair enough :smile: , fixed in d14320cef20eac47f88a0dc6194ea41f42366a6f

leloup314, 2017-11-03:
@DavidLP How should the std. track selection look like? In the [example](https://github.com/SiLab-Bonn/testbeam_analysis/blob/monopix/testbeam_analysis/examples/monopix_efficiency.py), the `selection_track_quality` is also a matrix like `align_duts` etc., right now its a slider. Should this also be changed?

leloup314, 2017-11-05:
@DavidLP @laborleben The interactive pre-alignment does not work in the GUI, even if it is started from the main GUI thread, because the `matplotlib` GUI starts a new `event loop` which does not work since there can only be one event loop. A `PyQt` sub-window for the interactive pre-alignment has to be implemented, I guess. Should I implement it to be also used in the script-version of `testbeam analysis`?

DavidLP, 2017-11-06:
> How should the std. track selection look like?

It should be an `n_duts x n_duts` matrix.
Fit DUTs should be tick box per DUT.

>Should I implement it to be also used in the script-version of testbeam analysis?

Interactive pre-alignment code is nevertheless very ugly. Rewriting it with better interface/calculation seperation would be good.

leloup314, 2017-11-06:
>It should be an `n_duts x n_duts` matrix.
What are the values for track selection? 0, 1 and 2 like [here?](https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/track_analysis.py#L175-L181). Should I implement an `OptionSpinBox` widget in which integer values from 0-2 can be selected and which can also be displayed in a matrix? 
>Fit DUTs should be tick box per DUT.
This is already done.

leloup314, 2018-02-19:
Plotting `track_finding` results into GUI consumes too much memory
[image]

DavidLP, 2018-04-23:
Congratulations! I close this issue, gui is stable enough to get one issue per bug.

closed 2018-04-23 by DavidLP

leloup314, 2018-04-23:
Yay! ;)