SiLab-Bonn/testbeam_analysis #73: Crash during map_cluster

leloup314, 2017-05-25:
[image]
[image] When running the `fei4_telescope` example script (above) and when executing the pre-alignment in the GUI (without multi-threading, just pooling to be safe, below) this error  
occurs and the analysis crashes.

DavidLP, 2017-05-29:
Do a `setup.py develop` to recompile the C++ parts to support the new data type.

closed 2017-05-29 by DavidLP

laborleben, 2017-05-29:
Optionally run `git clean -dxf` to remove all non-git-tracked files. Be careful, if you have changes in the pixel_clusterizer folder.

DavidLP, 2017-05-29:
> Be careful, if you made changes to the pixel_clusterizer folder.

So better do not do that ;-)

themperek, 2017-05-29:
Maybe do not use c but numba ;-)

DavidLP, 2017-05-29:
Numba also has quite some bugs and quirks (e.g. https://github.com/numba/numba/issues/1653). C is much more mature and if it crashes it is your fault. If Numba crashes it usually cannot be fixed by you and only circumvented ;-) Also C is twice as fast in real world applications.