SiLab-Bonn/testbeam_analysis #76: Additional culumns in merged file not addressed in find_tracks()

laborleben, 2017-06-17:
Commit 8de4f80 added new columns to the merged file, i.e. `n_hits_dut[dut number]`. Items of these columns are not copied in the `testbeam_analysis.track_analysis.find_tracks()` leaving `n_hits` at the wrong position in track candidates output file.

Moreover, the unit test files have zeros in these additional columns, which makes passing the test successful and leaving the bug undiscovered.

laborleben, 2017-06-17:
This is just a intermediate fix. We need probably a more flexible solution that copies the whole row and allows any additional column names. Additional column names might come from the clusterizer and might be necessary in a later analysis step, e.g. `result_analysis()`.

DavidLP, 2017-06-19:
All cython code unfortunately needs a fixed row definition. Thus flexible number and datatypes for columns needs rewriting of the cython code to numba.

commit 0945a11f (2017-06-19): BUG: sorting n_hits columns too, partially fixes #76
closed 2017-06-19 by DavidLP