SiLab-Bonn/testbeam_analysis #80: travis-CI is has timeouts

laborleben, 2017-07-14:
Message:
```
$ py.test --cov=testbeam_analysis

============================= test session starts ==============================

platform linux2 -- Python 2.7.13, pytest-3.1.2, py-1.4.34, pluggy-0.4.0

rootdir: /home/travis/build/SiLab-Bonn/testbeam_analysis, inifile:

plugins: cov-2.3.1

collected 34 items

testbeam_analysis/testing/test_alignment.py ...

testbeam_analysis/testing/test_analysis_utils.py .......

testbeam_analysis/testing/test_correlation_fixing.py .......

testbeam_analysis/testing/test_examples.py .

No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.

Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received

The build has been terminated
```

Maybe this can be solved with:
```
travis_wait py.test --cov=testbeam_analysis
```

See also: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received

DavidLP, 2017-07-18:
Try it, I have no time for this now

themperek, 2017-07-22:
Something is wrong with python 2.7.

laborleben, 2017-07-22:
Adding `travis_wait` helped to fix the timeout. But now the overall process takes too long. Unit testing alone take up to 45min with Python 2.7. The CI process is then terminated due to the following reason: `The job exceeded the maximum time limit for jobs, and has been terminated.`

Interesting because Python 2.7 takes much more time than Python 3.6.

DavidLP, 2017-07-24:
One can try calling `py.test --durations=10` during unit testing instead of `py.test --cov=testbeam_analysis` to show the 10 slowest unit tests and to check if my multithreading coverage hack makes troubles.

laborleben, 2017-07-26:
Fixed.

DavidLP, 2017-07-26:
Unfortunately not working. Sometimes travis hangs on python 2.7. Even 40 min timeout does not fix it.

DavidLP, 2017-07-26:
I checked python 2/3 under Ubuntu 14.04LTS 64-bit running `nosetests`. Both took exactly 219 s.
A wild guess it memory consumption and/or `numba` bugs that lead to a different behaviour under python 2.7.

DavidLP, 2017-08-30:
When running `nosetests` on travis`we use pretty much the available 4 Gb (or even 200 Mb above). This can lead to caching to disk decreasing run speed. 
[image]

When we switch to [sudo enable architecture](https://docs.travis-ci.com/user/reference/overview/) we have 8 Gb of RAM. Worth a try I guess.

DavidLP, 2017-08-30:
Looks like Python 3 just stay below the limit. That would explain the difference. Funny.
[image]

DavidLP, 2017-08-30:
With 8 Gb more RAM the [execution time is the same for Python 2/3](https://travis-ci.org/SiLab-Bonn/testbeam_analysis/builds/269948396?utm_source=github_status&utm_medium=notification). Nice, hidden bug. Fixed in GUI branch.

DavidLP, 2017-08-30:
Fixed in development

closed 2017-08-30 by DavidLP