SiLab-Bonn/testbeam_analysis #81: Sphinx has errors

laborleben, 2017-07-14:
Message:
```
[0K$ sphinx-versioning push -r development -w master -w development -b docs gh-pages .
[36m=> Reading config from docs/conf.py...[39m
Running Sphinx v1.6.3
loading pickled environment... not yet created
loading intersphinx inventory from https://docs.python.org/objects.inv...
intersphinx inventory has moved: https://docs.python.org/objects.inv -> https://docs.python.org/2/objects.inv

Exception occurred:
  File "/home/travis/miniconda/lib/python2.7/site-packages/sphinxcontrib/versioning/sphinx_.py", line 60, in builder_inited
    app.config.html_sidebars['**'] = StandaloneHTMLBuilder.default_sidebars + ['versions.html']
AttributeError: type object 'StandaloneHTMLBuilder' has no attribute 'default_sidebars'
The full traceback has been saved in /tmp/sphinx-err-eVymCp.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at <https://github.com/sphinx-doc/sphinx/issues>. Thanks!
Process ProcessWithCoverage-1:
Traceback (most recent call last):
  File "/home/travis/miniconda/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/home/travis/miniconda/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/travis/miniconda/lib/python2.7/site-packages/sphinxcontrib/versioning/sphinx_.py", line 219, in _read_config
    _build(argv, config, Versions(list()), current_name, False)
  File "/home/travis/miniconda/lib/python2.7/site-packages/sphinxcontrib/versioning/sphinx_.py", line 204, in _build
    raise SphinxError
SphinxError
[31m=> sphinx-build failed for branch/tag while reading config: <local>[39m
[33m=> Unable to read file, continuing with only CLI args.[39m
[36m=> Cloning gh-pages into temporary directory...[39m
[36m=> Building docs...[39m
[36m=> Gathering info about the remote git repository...[39m
[36m=> Getting list of all remote branches/tags...[39m
[36m=> Found: Hiterrors dbm development experimental gh-pages gui improvements master 0.0.1[39m
[36m=> Need to fetch from remote...[39m
[36m=> With docs: Hiterrors development gui improvements master 0.0.1[39m
[36m=> Passed whitelisting: development master 0.0.1[39m
[36m=> Root ref is: development[39m
[36m=> Banner main ref is: master[39m
[36m=> Pre-running Sphinx to collect versions' master_doc and other info.[39m
Running Sphinx v1.6.3
loading pickled environment... not yet created
loading intersphinx inventory from https://docs.python.org/objects.inv...
intersphinx inventory has moved: https://docs.python.org/objects.inv -> https://docs.python.org/2/objects.inv

Exception occurred:
  File "/home/travis/miniconda/lib/python2.7/site-packages/sphinxcontrib/versioning/sphinx_.py", line 60, in builder_inited
    app.config.html_sidebars['**'] = StandaloneHTMLBuilder.default_sidebars + ['versions.html']
AttributeError: type object 'StandaloneHTMLBuilder' has no attribute 'default_sidebars'
The full traceback has been saved in /tmp/sphinx-err-_gzLP0.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at <https://github.com/sphinx-doc/sphinx/issues>. Thanks!
Process ProcessWithCoverage-2:
Traceback (most recent call last):
  File "/home/travis/miniconda/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/home/travis/miniconda/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/travis/miniconda/lib/python2.7/site-packages/sphinxcontrib/versioning/sphinx_.py", line 204, in _build
    raise SphinxError
SphinxError
[31m=> sphinx-build failed for branch/tag: development[39m
Failure.

travis_time:end:02b2fefa:start=1500045983752368883,finish=1500046009786876393,duration=26034507510
[0Ktravis_fold:end:after_success.8
[0K
Done. Your build exited with 0.

No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received

The build has been terminated
```

DavidLP, 2017-07-18:
@themperek did this.

laborleben, 2017-07-18:
Related: https://github.com/Robpol86/sphinxcontrib-versioning/issues/39

themperek, 2017-07-21:
Fixed: https://github.com/SiLab-Bonn/testbeam_analysis/commit/9d41f70e143425bf1dfc3c96e411a9f0b75b3a99

closed 2017-07-21 by themperek