SiLab-Bonn/testbeam_analysis #102: High memory consumption due to SMC multiprocessing

leloup314, 2018-02-19:
When running the monopix analysis [script](https://github.com/SiLab-Bonn/testbeam_analysis/blob/monopix/testbeam_analysis/examples/monopix_efficiency.py) up to and including the `track_finding` step of the analysis, the memory consumption of `track_finding` reaches up to ~12 GB.

[image]
The figure above shows the memory consumption of the process ID connected to the analysis and the total memory consumption. I guess due to the multiprocessing, the memory is not assigned to the analysis PID, but clearly belongs to it (see figure below).

This effect can be avoided by setting `n_cores=1` in [`find_tracks`](https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/track_analysis.py#L127) (see figure below)
[image]

This was observed on:
- Ubuntu 16.04 (no programms running in background except 3 shells)
- Up-to-date `conda` environment with all packages from [here](https://github.com/SiLab-Bonn/pyBAR/wiki/Step-by-step-Installation-Guide)

Even though the run with `n_cores` set to 1 uses less memory, both runs take almost the same amount of time.

leloup314, 2018-02-22:
Possible bug in [line 182](https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/tools/smc.py#L182) of SMC: `chunk_size` of main process is given to every child process, causing every child process to consume the memory which the main process would use without multiprocessing. Possible solution is to set `chunk_size=int(self.chunk_size/self.n_cores)` in  [line 182](https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/tools/smc.py#L182) which distributes the total `chunk_size` of the main process among its children.
This then looks like this:
[image]
The black curve shows overall memory consumption of main process.

DavidLP, 2018-02-23:
The weird thing is that >> 1 Gb RAM is used per core in track finding since one chunk is only chunk_size * Track candidate size ~ 1000000 * 270 Bytes = 270 Mb. You have to memory profile per line to understand the underlying issue. I would not say that the issue is in SMC, since also the RAM per core is too high. Chunk size is meant per core, although that is not clear from the parameter description, that could be enhanced.

leloup314, 2018-02-23:
I managed to get a per line profiling:
[track_finding_profile.txt](https://github.com/SiLab-Bonn/testbeam_analysis/files/1751951/track_finding_profile.txt)

leloup314, 2018-02-23:
I managed to get a per line profiling:
[track_finding_profile.txt](https://github.com/SiLab-Bonn/testbeam_analysis/files/1751957/track_finding_profile.txt)

DavidLP, 2018-02-23:
Guess one also need to include `smc.py` in the profiling

DavidLP, 2018-04-23:
I also has troubles with track finding to exceed my 16 Gb of RAM. I reduced the chunk size to `100000` and RAM was constant at 4.4 Gb. So I guess that is a solution for now, although I am surprised of the overall memory consumption.

leloup314, 2018-04-23:
I guess `chunk_size=int(self.chunk_size/self.n_cores)` would be a more generic tmp. solution instead of having a static `chunk_size`. This would produce approx 4.4 Gb RAM usage independently of `n_cores`, right?

DavidLP, 2018-05-25:
Should be fixed now

closed 2018-05-25 by DavidLP