SiLab-Bonn/testbeam_analysis #107: Cython compile error during setup

leloup314, 2018-04-22:
After I did a fresh clone of `testbeam_analysis` and `conda update conda && conda update --all` I get the following error when doing `python setup.py develop` in the `testbeam_anaylsis` folder:

testbeam_analysis/cpp/analysis_functions.pyx:59:92: Invalid type.
Traceback (most recent call last):
  File "setup.py", line 30, in <module>
    Extension('testbeam_analysis.analysis_functions', ['testbeam_analysis/cpp/analysis_functions.pyx'])
  File "/home/leloup/miniconda2/lib/python2.7/site-packages/Cython/Build/Dependencies.py", line 1026, in cythonize
    cythonize_one(*args)
  File "/home/leloup/miniconda2/lib/python2.7/site-packages/Cython/Build/Dependencies.py", line 1146, in cythonize_one
    raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: testbeam_analysis/cpp/analysis_functions.pyx

laborleben, 2018-04-22:
Use 
```from testbeam_analysis.cpp.data_struct cimport numpy_hit_info, numpy_cluster_info```
instead of
```from data_struct cimport numpy_hit_info, numpy_cluster_info```
in https://github.com/SiLab-Bonn/testbeam_analysis/blob/development/testbeam_analysis/cpp/analysis_functions.pyx.

Does it solve your issue?

leloup314, 2018-04-22:
Yes, this solves the issue!

laborleben, 2018-04-23:
OK, I'll prepare a PR.

commit dc47e8d7 (2018-04-23): REG: fix cython compiler issues, fixes #107
closed 2018-04-23 by DavidLP