SiLab-Bonn/testbeam_analysis #113: Track fitting tab

coquelin77, 2018-04-23:
It is not clear what is on x and what is on y for `Selection hit DUTs` and  `Selection fit DUTs`

leloup314, 2018-04-29:
Is this still not clear? If so, any suggestions on how to improve this?

DavidLP, 2018-04-29:
A hint when mouseover could be useful

leloup314, 2018-04-29:
We already have the descriptions from the docstrings for each option when hovering over the name of the respective option. Usually this also contains an example. I don't really know what to add here. I mean the 'y' axis always contains the names of your DUTs which you either gave them or 'Tel_i' as default. For `x` axis in e.g. `track_fitting` it says `Fit Tel_0`. I could make a tooltip saying "Fit tracks for Tel_0".

YannickDieter, 2018-04-29:
I agree. The word „fit“ should already tell the user that it means that the selection is for tracks fitted for the respective DUT.

leloup314, 2018-05-01:
I'm closing this since I don't see a need for any changes. Feel free to reopen if you really want to change anything here.

closed 2018-05-01 by leloup314