SiLab-Bonn/testbeam_analysis #121: PyQt5 - missing requirement

owtscharenko, 2018-09-06:
PyQt5 is needed for gui, but  not installed upon installation of testbeam_analysis. It is also not part of the requirements.txt

leloup314, 2018-09-06:
I always assumed this was one of `matplotlibs` dependencies but I just figured its not. So we should add `pyqt` to the `requirements.txt` This should install PyQt version 5

laborleben, 2018-09-06:
I want to make you aware that the issue with PyQt package is more complicated. The name `pyqt` is just package name for PyQt Version 5 in the Anaconda/Miniconda universe, whereas in the pip universe the name is `PyQt5`. Choosing one universe will break the installation requirements for the other. There is no satisfying solution available. Check out this link: https://github.com/uvemas/ViTables/issues/79

laborleben, 2018-09-07:
@DavidLP Thanks for the second link. Maybe you should also post this to 
https://github.com/uvemas/ViTables/issues/79.